bri3d/VW_Flash #152: VW Flash unlock issue

JBVanquish, 2025-01-03:
Hey, I attempted to unlock my ECU using VW Flash, however the process froze when it got to flashing block 3. When I retry the unlock process it doesn't go past the "Clearing DTCs" stage.
[flash.log](https://github.com/user-attachments/files/18297949/flash.log)
[flash_details.log](https://github.com/user-attachments/files/18297951/flash_details.log)
[udsoncan.log](https://github.com/user-attachments/files/18297953/udsoncan.log)

bri3d, 2025-01-03:
Please try downgrading to 0.6.0 and see if that gets you sorted out.

JBVanquish, 2025-01-03:
Thanks, I'll give it a try. Do you have any advice on how I can recover the bricked ECU?

bri3d, 2025-01-03:
It’s very unlikely to be bricked based on the logs you sent. Try again with
0.6.0. I think there may be an issue with recovering an ECU stuck in CBOOT
in 0.7.x, caused by some issue with the J2534 interface not resetting
properly if there is no response to the OBD “04” request to clear DTCs.

JBVanquish, 2025-01-03:
That definitely did the trick. I managed to flash the unlock patch, as well as the tuned file. I seem to have developed a new DTC after the flash, "P0068 MAP/MAF throttle position correlation". Any idea what may be causing this?