bri3d/VW_Flash #157: Simos 12.1 CAL Data flash

TheFlashBold, 2025-02-13:
Simos 12.1 CAL Data stuck in CBOOT after flash.
I updated to FL_8V0906264E__0003 via frf, which was fine.
When modifying CAL Data and flashing it, my ecu is stuck in CBOOT.
Reflashing the original CAL Data fixes it.
Since VW_flash updates the checksum, this should work and i dont need to unlock the ecu right or do I misunderstand something?

bri3d, 2025-02-17:
No, there are two parts, checksum and signature checking. Simos12.1 has signature checking so it would need to be unlocked to disable the signature checking. I don't have an unlocking patch made for Simos12.1. I believe you could do it the same way I did for Simos18 but there are also simpler exploits available for Simos12 (I believe you can sneak a CBOOT which has been marked as valid into the CAL area and it won't be sig checked again, for example)

TheFlashBold, 2025-02-17:
Thank you.
Any resources where I could start and how to modify the CBOOT?
When opening the bin in ghidra with tc1979 definitions, it doesn't seem right.

TheFlashBold, 2025-02-28:
Couldn't we fix the signature? Since the ECU checks it, it has to be reversable right?
Or any idea how to "sneak a valid marked CBOOT"?