bri3d/VW_Flash #160: Incorrect FRF Flashed and Can't Revert

hybridr, 2025-02-20:
I wanted to update my S3's ECU software version from 8V0906259H_0001 to 8v0906259K_0003 based on some TSBs I've read about. Unfortunately, I mistakenly flashed 8V0906264K_0003. I am not able to flash it back with VW Flash. It skips over after extracting the FRF. Is there any way I can do this? I tried the command line version with Python and it tells me the S50 file is incompatible with SG1.

[flash.log](https://github.com/user-attachments/files/18895561/flash.log)
[flash_details.log](https://github.com/user-attachments/files/18895563/flash_details.log)
[udsoncan.log](https://github.com/user-attachments/files/18895562/udsoncan.log)

TheFlashBold, 2025-02-21:
it stops here
2025-02-20 14:21:35 [INFO] SimosFlashHistory: <VIN> Connected: Flashing blocks: [1, 2, 3, 4, 5]
2025-02-20 14:21:36 [ERROR] SimosFlashHistory: Server sent an invalid payload : b'\x7f>\x13'

TheFlashBold, 2025-02-21:
It fails to "Checking programming precondition, routine 0x0203..."

aarons3, 2025-02-21:
immo bricked, will need open it up and boot it, flash correct sw back on.