bri3d/VW_Flash #161: DQ500 0DL Flash Support

flamingtoast123, 2025-03-15:
Hello, is it possible Bosch DQ500 DSGs will be supported in the future, as I believe they're very similar to the DQ381?

Massive respect for making this software open source, successfully flashed my Mk2 Tiguan in Australia.

threepotMR2, 2025-05-01:
Yes please, DQ500 would be very helpful, I would love this too :)

These gearboxes are used in all sorts of conversions, DQ500 from diesel Tiguan and Q3 especially.

otnep, 2026-10-16:
Hi all,

I've been digging into the stock FRF for my DQ500 (0DL300013G, SW 2620). I haven't flashed anything yet, but I think I can see why @evo77t's TCU died at "Requesting Download for Block 3".

The TheFlashBold fork runs `--dq500_0dl` through `dsg_flash_utils`. The block addresses, AES key and SA2 are correct, but the file prep doesn't match what's in the stock ODX:

- The stock ODX uses data format **0xAA** for RequestDownload. The dsg path sends 0x11. If the TCU rejects that right after erasing block 3, you'd get exactly that brick. Without logs this is only my guess, though.
- The CAL checksum is a normal CRC32 over the range given at +0x38/+0x3C, stored big-endian at **+0x44** (stock value is `3D45F91E`). The dsg code writes a JAMCRC into the last 4 bytes instead.
- The per-block CRC the loader checks is the ODX `FW-CHECKSUM`, a plain CRC32 of the whole decompressed block. The dsg path uses `FFFFFFFF` from the profile.
- Padding: the original stream is padded to 16 bytes with zero-length LZSS tokens, not zeros. Upstream `lzss.c` recompresses the stock CAL byte-for-byte identical to the original, so the codec itself is fine.

Side note: the CVN (OBD 09/06) is just the ASW CRC plus the CAL CRC (both at +0x44). For stock that's `553CE495`, which matches the `ASWCAL_CHECKSUM` comment in the ODX and what my car reports.

Now the part I need help with. I've made a small CAL-only change: two thresholds for the idle-speed-raise request. I repacked it into the stock FRF this way. BOOT and ASW are byte-identical to stock, and all the CRCs check out. I'd like to flash it with ODIS-E, but I don't have a bench tool, so if it goes wrong I have no way back.

Does anyone with a bench backup (KT200, Flex, Autotuner…) want to try it first? I won't post the firmware, but I can share a script that patches and checks your own stock FRF. Even a test on a different 0DL software version would help. What I really want to know is whether CBOOT accepts a modified CAL when only the CRCs are fixed.

Thanks!

TheFlashBold, 2026-10-16:
Thanks for checking @otnep, I changed it to 0xAA for the DQ500 0DL in my fork https://github.com/TheFlashBold/vw_flash!
Would be great if someone with a recovery option could test this.