bri3d/VW_Flash #163: MY24 ECU Support [SIMOS18.10 - 81A906259B - SCQ Bootloader]

CherryTuning, 2025-04-19:
Hello,
I'm trying to get VW Flash working with a 2024 Audi SQ2 with SIMOS18.10

VW ECU Hardware Number : 06K907425M 
VW ECU Hardware Version Number : H80
VW Spare Part Number : 81A906259B 
VW Application Software Version Number : 0003
Boot Loader Identification : SCQ__ CB.00000130 004.00 SCQ 
VW System Name Or Engine Type : R4 2.0l TFS  
Engine Code Letters : DNFC

I do notice when running the identification process, I get 6x the same error, stating:
Block fingerprint dated '23 December 15 does not appear to be from SimosTools or VW_Flash.

I am also unable to use the FRF Extraction tool, where I'm getting the error:
"Cannot unpack non-iterable NoneType object"

I've got a matching FRF file for this part number and software version, and I'm going to persist with unpacking the FRF (I've been able to get the ODX out of it so far using ODIS) - but I'm concerned I may cause damage to my (essentially) brand new car if I persist despite errors.

Is this ECU and bootloader supported?

aarons3, 2025-04-19:
That appears to be simos19.3

CherryTuning, 2025-04-21:
> That appears to be simos19.3
> […](#)

Interesting, I'll have to physically pull the ECU to confirm, when I google the numbers it comes back as both 18.10 and 19.3 - but I'm inclined to suggest you're right

lndmnn, 2025-09-03:
SCQ is the Application Identifier for Simos 19.3. You can totally forget to do anything with it and VWFlash.