bri3d/VW_Flash #167: Bench recovery

kamold, 2025-05-01:
Firstly, wanted to commend you on an amazing project!

I have a couple of 18.1 ECUs that I've collected from workshops which have suffered from an incorrect flash that I'm interested in recovering. 
I have a Raspberrypi and CAN FD HAT, got everything installed and confirmed CAN comms via loopback etc.  I'm using a BDM frame and probes, which I use for reading/writing to emmc flash in circuit on navigation systems.

Have confirmed the ECUs are responsive by connecting to another tool which can ID them.

When I start upload in the bootloader.py script, it is not able to proceed.  99% sure it's a fault on my side (have checked all the probe locations, HAT pinouts, level shifter output voltages etc until I'm blue in the face).

So my question is - is there any logging or debug function that will assist with pointing me in the right direction as to where I'm going wrong?

aarons3, 2025-05-01:
It's been a hot minute since I've used this, but IIRC when that command is run, first the ecu is reset, and then the HWCFG / BSL pin is brought to gnd.

I would start by verifying both GPIOs used for these are functioning as intended.  I had an issue with this and it was linked to some of the GPIOs not coming all the way down to ~0V, and it wouldnt enter BSL mode.  Make sure the grounds are tied together, and test out various GPIOs if needed.  In my case, IIRC I ended up moving BSL to GPIO 6 or something and it worked much more consistently.

kamold, 2025-05-01:
Thanks so much for your reply I'll follow your suggestions and report back.