bri3d/VW_Flash #169: Flashing Stock VW to Stock VW

jmkizer, 2025-06-06:
Just a thought here.  For those who wish to take a stock VW software load and simply update it to the latest factory code could their not be a simplified flash procedure?  I was thinking is it required to unlock the ECU or could one simply pick Flash Stock (re-lock ECU) and upgrade the factory firmware from one version to another?

bri3d, 2025-06-06:
>  could one simply pick Flash Stock (re-lock ECU) and upgrade the factory
firmware from one version to another

Yes, do this and pick an FRF. Make sure that the upgrade has compatible
immobilizer data so you don’t make a brick. That’s all there is to it.

jmkizer, 2025-06-06:
Some thoughts:

1.  What would constitute / trigger a lock based on immobilizer data?  For example. I am looking to update a US -> US, same ECU HW PN, same ECU SW PN, and one revision up.  i..e 3CN906259_0005 to 3CN906259_0006.  I couldn't imagine a immobilizer lock based on this.
2. Could we not add a warning in the GUI that would check for an incompatibility in the files that MAY trigger and immobilizer lock?  Perhaps the next version?  Additionally, we could change the text of the buttons/selection options to make this a clear path of upgrading ECU files without having to unlock.

closed 2025-11-20 by bri3d