bri3d/VW_Flash #172: Patching Simos 18.10 for HSL logging

Mikitaii, 2025-08-01:
Hello everyone,

I tried to patching my file FL_5G0906259Q__0007 using as reference [this](https://github.com/bri3d/VW_Flash/blob/878b4831fc9e77d01de73efb1472b787d2025036/docs/patch_1810.bin) example. Description provide FL_5G0906259Q__0005 file but comparing it the difference only in CAL that not use regarding this method. I added code into two places but after flash car didn't start with many errors at dashboard, As I got it refer for wrong CS. I checked it by myself but everything correct. 
Can't understand where is mistake. Maybe someone can help me to change file based on 0007 version or share worked example for simos 18.10?

Samsara69, 2025-08-02:
Hello dear, sorry no answer but question.
I planned to use this tool for 18.10 read/write (chiptuning) ,do you know if it can without problem?

Mikitaii, 2025-08-04:
> Hello dear, sorry no answer but question. I planned to use this tool for 18.10 read/write (chiptuning) ,do you know if it can without problem?

Hello,

I've used it in such way since last year with simos 18.10. No issue for that task. Unlock/write/logging for easy peasy.

closed 2025-11-20 by bri3d