bri3d/VW_Flash #174: LZSS Compression and Encryption of Flash Data

jakka351, 2025-08-09:
I am trying to flash a Simos ECU, and am wondering about the LZSS compression and AES encryption that is used on the flash data bytes before they are sent to the ECU. Using 0x34 requestDownload, I can see in CAN logs that several flags are set in the compression/encryption bytes for this message. Will the ECU accept uncompressed and unencrypted data if you set these flags in the requestDownload message to 0, or is compressed and encrypted flash data required by the ECU? 

Thanks

jakka351, 2025-09-01:
Was successfully able to flash a Simos 18.1 ECU with uncompressed and unencrypted flash data (request download flag 0x00), and successfully able to flash with request download flag 0xA0, compressed data with no encryption. Am yet to test flashing flag 0xAA compressed and encrypted flash data.

Pakitkoo, 2025-09-04:
good job friend

closed 2025-10-29 by jakka351